SpiNNakerManchester/SpiNNFrontEndCommon #793: host and machine compressors using different configs

Christian-B, 2021-04-26:
Host compression uses:
router_table_compression_target_length

machine compressors use:
router_table_compress_as_far_as_possible

This is especially weird when the MachineBitFieldRouterCompressor fails on machine and then falls back to host

Suggestion nuke: router_table_compression_target_length as a cfg

rowleya, 2021-04-26:
Both of these were part of the algorithm because it was being used for research.  In practice, unless a downstream algorithm is going to need more compression (none currently do), compressing only as much as is needed is likely to work.

Christian-B, 2021-04-26:
I fully agree the whole set of button and dials added to control compression only make sense in research/ testing.
In all other cases stop as soon as it fits makes sense.

Which is why it makes sense to reduce the number of dials to just one. If doing research continue compression as far as possible.

closed 2021-06-24 by Christian-B