SpiNNakerManchester/SpiNNFrontEndCommon #848: Split AbstractSpinnakerBase into model(s), views(s), controllers(s)

Christian-B, 2021-09-14:
Currently AbstractSpinnakerBase and  its children include
1. Data
2. methods to access data
3. methods to run algorithms
4. methods for users
5. methods for alogirthms
ect

Idea wuld be to split this into several smaller separate classes

Christian-B, 2021-09-21:
Together with https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/issues/849 it can be considered which data needs to go into the model and what is just pass through bwteeen algorithms

dkfellows, 2023-09-27:
How much of this is still relevant?

Christian-B, 2023-09-28:
With the data objects being moved to the views this is a lot less import.

As the software team will be significantly reduced do to the end of the HBP project this is now a "Will not do"

closed 2023-09-28 by Christian-B