SpiNNakerManchester/SpiNNFrontEndCommon #866: Energy Report vs multiple _do_run

Christian-B, 2021-09-21:
The energy report is only really designed for a single _do_run clcyle

The two options are.

1. Only run the Energy report stuff on the first _do_run cycle
2. Nuke the whole energy report as noone uses it
3. Fix the Energy Report

andrewgait, 2021-09-21:
I’d much rather keep this if we can - and if people aren’t using it then we should be directing them to do so.

Christian-B, 2021-09-21:
Yes I do see the point in keeping the Energy Report.

Is it worth upgrading it to mutliple _do_run cycles or is the information from a single do_run enough?

andrewgait, 2021-09-21:
I think it's worth upgrading to multiple do_run cycles, though whether you can distinguish at this level between multiple calls to run() and auto-pause and resume ought to be considered.

Christian-B, 2021-09-21:
In master the report runs every do_run cycle regardless if it is and auto pause or mutliple sim.run.

Master also does not reset timings on a hard reset. It should.

The time execute_time depends on when the executor decides to run FinaliseTimingData

Christian-B, 2021-09-21:
With the no executor stuff it is worth reconsidering the phases reported.

Christian-B, 2021-09-21:
What stages o report is also related to https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/issues/849

Christian-B, 2021-09-21:
Care is need to make sure _extraction_time and run_time do not both include buffer extraction.

Christian-B, 2021-09-27:
Also looks like the EnergyReport goes boom on runtime is None

So either fix that or add a condition to run the report that runtime is not None

Christian-B, 2021-10-05:
With the removal of the pacman executor and _pacman_provenance the engery report is seriously broken so currently (in no executor work) deactivated.

As provenance is being changed in https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/issues/867 it is not worth fixing the "CHEAT!" in ComputeEnergyUsed until then,

Christian-B, 2021-10-12:
Turn back on spynnaker_integration_tests/test_power_monitoring

Christian-B, 2022-01-21:
the energy report also completely ignores the get machine time

Christian-B, 2022-01-21:
I also notice that buffer_manager adds extraction_timing even when it is inside the run loop.

That may be due to BufferExtractor being move during the removal of the pacman exectuor

closed 2025-01-03 by Christian-B