SpiNNakerManchester/SpiNNFrontEndCommon #867: Write Provence straight to database or file

Christian-B, 2021-09-21:
So two things to consider.
1 Do we want to support xml, json and sql provenance of just sql?

2. Is there any reason to keep a ProvenanceDataItem objects in lists or should we just write provenance as soon as possible.

rowleya, 2021-09-21:
I personally prefer the sql; conversion from sql to xml is reasonably easy to do if someone requests it.

There is no reason to keep the the items in a list in general; this was just done to allow something else to use it, but just as easy to use things directly from the sql.

Christian-B, 2021-09-22:
Will require reworking the RedundantPacketCountReport

Christian-B, 2021-09-22:
Writing provenance ASAP will avid having to handle provenance in exception handling which leads to little tested code tha can hide bugs such as https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/pull/869

Christian-B, 2021-09-27:
also look at BitFieldCompressorReport

Christian-B, 2021-09-27:
Will need to look at GraphProvenanceGatherer, PlacementsProvenanceGatherer, RouterProvenanceGatherer

Is it better to generate provenance this way or as the data is available

Christian-B, 2021-10-05:
When this and https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/pull/847 consider adding provenance to FecTimer so that ComputeEnergyUsed has the data it needs