SpiNNakerManchester/SpiNNFrontEndCommon #868: better control C error handling

Christian-B, 2021-09-21:
With the no executor work it may be work having another look at exit handling and control c.

For example @dkfellows suggests using https://docs.python.org/3/library/atexit.html

dkfellows, 2022-04-29:
Stuff that goes in an `atexit` handler should be things that _must_ be done on close. (Flushing IO buffers is the classic example, which Pythkn does for us.) The main things for us are going to be shutting down the current machine allocation controller, and ensuring that we have closed all IO channels that we're writing to.

Cleaning up memory should not be done; the OS is far better at that. Nor should we interact with the user, as they've already told us to go away.

rowleya, 2022-05-03:
There are two points at which exit could occur which might require some clean-up:
1. After run has been called but before it has finished.
2. After run has returned but before ```end()``` is called.

Exit might be caused by an error, or by a user pressing ctrl-c; both of these are an Exception in python so likely can be handled in the same way.