SpiNNakerManchester/SpiNNFrontEndCommon #872: cfg flags for compressor reports

Christian-B, 2021-09-27:
The current status is write_routing_table_reports turns on off all compressor reports
including unCompressedRoutingTableReport

Then write_routing_tables_from_machine_reports
adds but only if the above is also True
 ReadRoutingTablesFromMachine
compressedRoutingTableReports
comparisonOfRoutingTablesReport
CompressedRouterSummaryReport
RoutingTableFromMachineReport

Do we want to give each report its own cfg?

ReadRoutingTablesFromMachine can then be added if any of these is set.

rowleya, 2021-09-27:
That makes a lot of sense.  I am not sure what the difference is between comparisonOfRoutingTablesReport and CompressedRouterSummaryReport but if both make sense separately, a separate config also makes sense.

Christian-B, 2021-11-12:
tricky bit is if you have different cfg flags how to work out when to run ReadRoutingTablesFromMachine

Options include
1. giving ReadRoutingTablesFromMachine a list of cfg flags

2. giving each report function a call back to get the compressed if not yet in host

rowleya, 2021-11-12:
I would suggest that having a function to get the routing tables from the machine might make more sense (i.e. basically 2).  This means that the data is obtained only when needed.  I suppose if you then store the routing tables from machine somewhere too, these can be passed in if available before the report is called, and if None, do the read then.