SpiNNakerManchester/SpiNNFrontEndCommon #891: RouterCollisionPotentialReport Issues

rowleya, 2021-11-12:
RouterCollisionPotentialReport appears to be enabled by default where it wasn't before, and has no config to stop this.  When run with virtual chips, I seem to get the following error:
```
INFO: Router collision potential report exited with AttributeError after 0:00:02.619424
Traceback (most recent call last):
  File "/home/zzalsar4/git-local/TestScripts/test_input_network.py", line 174, in <module>
    p.run(run_time)
  File "/home/zzalsar4/SpiNNaker/sPyNNaker/spynnaker8/__init__.py", line 675, in run
    return __pynn["run"](simtime, callbacks=callbacks)
  File "/home/zzalsar4/SpiNNaker/lib/python3.8/site-packages/pyNN/common/control.py", line 111, in run
    return run_until(simulator.state.t + simtime, callbacks)
  File "/home/zzalsar4/SpiNNaker/lib/python3.8/site-packages/pyNN/common/control.py", line 93, in run_until
    simulator.state.run_until(time_point)
  File "/home/zzalsar4/SpiNNaker/sPyNNaker/spynnaker8/spinnaker.py", line 88, in run_until
    self._run_wait(tstop - self.t)
  File "/home/zzalsar4/SpiNNaker/sPyNNaker/spynnaker8/spinnaker.py", line 119, in _run_wait
    super(SpiNNaker, self).run(duration_ms, sync_time)
  File "/home/zzalsar4/SpiNNaker/sPyNNaker/spynnaker/pyNN/abstract_spinnaker_common.py", line 289, in run
    super().run(run_time, sync_time)
  File "/home/zzalsar4/SpiNNaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 877, in run
    self._run(run_time, sync_time)
  File "/home/zzalsar4/SpiNNaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 1025, in _run
    self._do_mapping(total_run_time)
  File "/home/zzalsar4/SpiNNaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 2218, in _do_mapping
    self._report_router_collision_potential()
  File "/home/zzalsar4/SpiNNaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 2110, in _report_router_collision_potential
    report(self._routing_table_by_partition,
  File "/home/zzalsar4/SpiNNaker/SpiNNFrontEndCommon/spinn_front_end_common/utilities/report_functions/router_collision_potential_report.py", line 37, in __call__
    collision_counts = self._generate_data(
  File "/home/zzalsar4/SpiNNaker/SpiNNFrontEndCommon/spinn_front_end_common/utilities/report_functions/router_collision_potential_report.py", line 67, in _generate_data
    other_chip_x = chip.router.get_link(link).destination_x
AttributeError: 'NoneType' object has no attribute 'destination_x'
```

rowleya, 2021-11-12:
Note that the error might be down to something I have changed in my branch

Christian-B, 2021-11-15:
That RouterCollisionPotentialReport needs a cfg 100% agree.

Christian-B, 2021-11-15:
The error however does not look like router_collision_potential_report fault but rather a GIGO

router_tables_by_partition is saying that a link exists that the chip can not find

Christian-B, 2021-11-15:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/pull/895 as cfg.

HIGHLY suggest @rowleya turns it on and runs TestScripts/test_input_network.py

Christian-B, 2022-07-14:
report still uses n_keys_map

Christian-B, 2022-07-19:
@andrewgait notes: can we rename this report to change "protential" to "potential"

Christian-B, 2023-05-25:
Report has now been removed

closed 2023-05-25 by Christian-B