SpiNNakerManchester/SpiNNFrontEndCommon #896: ReverseIPTagMulticastSourceMachineVertex  keys_per_timestep

Christian-B, 2021-11-18:
ReverseIPTagMulticastSourceMachineVertex._recording_sdram_per_timestep

had:
keys_per_timestep = math.ceil(
         (receive_rate / (machine_time_step X 1000.0)) * 1.1
)

Now the same but soon using simulation_time_step_us

should this not be 

keys_per_timestep = math.ceil(
         (receive_rate / (machine_time_step / 1000.0)) * 1.1
)

Which is now
keys_per_timestep = math.ceil(
         (receive_rate / simulation_time_step_ms) * 1.1
)

Christian-B, 2021-11-19:
I changed this in the global data branch but left a comment to this issue.

rowleya, 2021-11-19:
The receive_rate I would guess was meant to be in Hz, so if it works out keys_per_timestep correctly, it makes sense to update it.

Christian-B, 2021-11-19:
Hz is units per second.

In which case the default rate of 10  would be 10 spikes per second which is rather low.

Are we sure the rate is not in 

mHz  millihertz 

--- 
If the rate is in Hz then the correct is
keys_per_timestep = math.ceil(
(receive_rate / simulation_time_step_s) * 1.1
)
which at defaults is 
ciel(10 / 0.001 * 1.1) which is 1

if it is millihertz 
keys_per_timestep = math.ceil(
(receive_rate / simulation_time_step_ms) * 1.1
)
defaults to
ceil (10 / 1 * 1.1) which increases the result to 11

rowleya, 2021-11-19:
It is hard to say what a good "receive_rate" value is in these cases as it depends on what the user is doing.  Ideally the user would set this anyway if they really want recording (and documentation is then the best option, or else make it required if the user records a SpikeInjector).  I think the value should be considered to be in Hz as that is the PyNN standard unit anyway.

I think the 1.1 multiply is probably unnecessary and confusing at best.

rowleya, 2021-11-19:
Oh and note that the value to be calculated is keys_per_timestep which is calculated from receive_rate, so:
```keys/ts = (keys/sec) / (ts/sec) = (receive_rate) / (1/simulation_time_step_s)```

Where```ts/sec``` = timesteps per second (rather than seconds per timestep).