SpiNNakerManchester/SpiNNFrontEndCommon #988: write_provenance_data cfg flag never used.

Christian-B, 2022-10-11:
Found in both spynnaker.cfg and spynnaker.cfg.

[Reports]
write_provenance_data

But never read.

Should this be used to turn of chip iobuff extraction of just nuked?

Christian-B, 2022-10-11:
There is the flag read_provenance_data but that is used for many other things too.

andrewgait, 2022-10-11:
Isn't there also a flag called extract_iobuf as well?

Christian-B, 2022-10-11:
> Isn't there also a flag called extract_iobuf as well?

No read_provenance_data control that too.

andrewgait, 2022-10-11:
[Reports]
extract_iobuf = False

If you don't set this to True, you don't get any iobuf...

rowleya, 2022-10-11:
There was a point where we wanted to report errors but not write provenance as it was filling up disk space I think.  I think this might be less of an issue with the database but not sure; happy to remove that option until asked for again though!

extract_iobuf is normally off I think to avoid reading files no-one cares about, but I do think that still works separately from provenance in master.  I don't know if this is about to change though...

Christian-B, 2022-10-11:
I stand corrected. Yes there is an extract_iobuff setting and yes it is off by default!.
But IF read it is always written even if write_provenance_data is False as that one is never checked.
I personally this that if read then yes always write so nuke the unused cfg flag.

read_provenance_data afffects the profile_data_gatherer which also writes to the same directory as the iobuff extractor.

andrewgait, 2022-10-11:
In that case I agree, nuke write_provenance_data.