SpiNNakerManchester/SpiNNFrontEndCommon #1198: Python 3.13 depends on a scipy source build

Christian-B, 2024-07-10:
ref: https://github.com/scipy/scipy/issues/16308
"""
you need to have openblas installed if you want to build scipy from source, which is what your log indicates. Which makes sense: scipy 1.14.0 on pypi, understanable, does not have python 3.13 wheels. So you'll need to either build scipy from source (which requires openblas etc) or use nightlies from https://anaconda.org/scientific-python-nightly-wheels/scipy/files (usual caveates about nightly builds apply)
"""

Christian-B, 2024-07-10:
I do not see supporting pre release 3.13 important enough to try to build scipy from source.
Making this for a later release

EwoutH, 2024-07-29:
Tomorrow the Python 3.13 release candidate will be released, which makes 3.13 ABI stable and allows (and encourages) wheels to be uploaded to PyPI.

Christian-B, 2024-09-02:
Scipy now supports Python 3.13  (and probably has since July 30)

There is a small Typing clash.

spinn_front_end_common/utilities/iobuf_extractor.py:19:0: E0611: No name 'Pattern' in module 'typing' (no-name-in-module)

Suggstion is to not support Python 3.13 until adapting the code for https://peps.python.org/pep-0585/