SpiNNakerManchester/SpiNNFrontEndCommon #1205: Remove data once uploaded

rowleya, 2024-08-22:
Once data has been written to the machine it gets left in the database.  This can grow quite large and I don't think it gets re-used.  It would therefore be useful to remove the data as it gets uploaded.  This could either be done region-by-region (ideal as quickly removes things) or at the end of the uploading (probably easier).

Christian-B, 2024-09-16:
I completely DISAGREE!

The provenance aspect of having the data around if something goes wrong it much more important than saving a little disk space.

The time taken to delete the data is highly likely more than the time lost by the heavily optimised sqlite3 having a larger database.

Suggest closing will not do!

Christian-B, 2024-09-16:
Or at the very most a simple nuke the WHOLE ds.sqlite3 once it has been used.