SpiNNakerManchester/SpiNNFrontEndCommon #1233: BufferManager _send_initial_messages weirdly handles regions to small

Christian-B, 2024-11-07:
The error message claims to warn it is out of space but the checks do otherwise.

1. Dettermine how to act if vertex.is_next_timestamp(region)  gives more data than  vertex.get_region_buffer_size(region) said it should

2. Implement that

Christian-B, 2024-11-07:
If we error on too much data the SSA models in run forever should only send get_max_run_time_steps of data

rowleya, 2024-11-07:
In some ways these are supposed to be independent.  The get_region_buffer_size is supposed to give you the size you want the buffer to be.  The calling code is then supposed to keep adding new things until that gets full and then stop asking.  If the calling code keeps asking even when the buffer space is filled that is a problem in the calling code...

This was all designed to work so that buffers could be dynamically filled during run, but we don't do that any more, instead relying more on the pause and resume system for this to happen.  This might mean that the functions are not quite right; for example it does seem odd that we ask the vertex what size the buffer region is, but maybe we should be telling it (or using the space that should be available with auto pause and resume)?

rowleya, 2024-11-07:
In run_forever, the SSA would have to stop sending once the space is exhausted.  This is "a bit odd" in that if you have spikes to send and there is no space, they will never get sent.  This might be considered an error therefore, or at best a warning that some spikes won't be sent in that mode.  If you are using an SSA which has specific requirements, run_forever might just not be a good option.