SpiNNakerManchester/SpiNNFrontEndCommon #1264: Upload log database to spalloc when using

rowleya, 2025-02-07:
Once spalloc supports uploading of data (see SpiNNakerManchester/JavaSpiNNaker#1217), add a call here to do the upload

Christian-B, 2025-02-07:
Ideally the upload would happen after the create job request has been started.

There is time between that and when the job hs been allocated boards and these are ready.

rowleya, 2025-02-07:
In some ways it doesn't matter; once a job has been established, the log database can be attached regardless of when the boards are actually allocated.  The log data will just be stored ready for use when requested; this won't be possible until the board is at least booted anyway, and won't be useful until something has actually run and written to IOBUF of course!